RocketPy-Team/RocketPy #564: ENH: Allow saving plots instead of showing them

nalquas, 2024-02-27:
**Is your feature request related to a problem? Please describe.**

All matplotlib plots generated by RocketPy (for example via [`Rocket.draw`](https://docs.rocketpy.org/en/latest/reference/classes/Rocket.html#rocketpy.Rocket.draw) or [`flight.all_info`](https://docs.rocketpy.org/en/latest/reference/classes/Flight.html#rocketpy.Flight.all_info)) are shown using matplotlib's [`show`](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.show.html) function. This does not really allow for automatic export of these plots (as is the case when not using Jupyter Notebook).

**Describe the solution you'd like**

Add a parameter to the info functions (and other plot-generating functions) for selecting between immediately showing the plot and saving the plot to a file. Internally, this can be done using matplotlib's [`savefig`](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.savefig.html) function.

Gui-FernandesBR, 2024-02-27:
Hello @nalquas , thank you for opening this issue, that's a wonderful feature request.

Please let us know if you would be interested in contributing to the solution.

Here are some suggestions that might help:
- Hackathon 2022: https://github.com/RocketPy-Team/RocketPy-Hackathon-2022/pull/104/files#diff-2a21c44703e31fe4922b5045115a85bed731e90a3e622139125a74a4bcf120d9
- Hackathon 2020 (private repo): https://github.com/RocketPy-Team/Hackathon_2020/pull/88/files

Overall, allowing the user to save the plots instead of just plotting them is quite helpful, specially to teams wanting to create reports for competitions submissions.

Gui-FernandesBR, 2024-04-09:
> Hi @Gui-FernandesBR, thank you for the response. I would be interested in working on this (I am part of [a student team](https://www.bears-space.de/) that needs this feature), although I see you self-assigned the issue in the meantime. Did you already start working on it?

No I have not started. Feel free to implement it by yourself, asking any questions if needed. We would be extremely glad receiving a PR from you.

closed 2024-11-10 by Gui-FernandesBR