RocketPy-Team/RocketPy #600: ENH: Use eng files to define additional motor parameters

chasgior214, 2024-05-15:
**Is your feature request related to a problem? Please describe.**

It would be nice to not have to write out motor parameters that are already defined in an eng file when using an eng file to supply the thrust curve for a motor. 

#309 addressed automatically adding the burnout time, but it did not implement that for the GenericMotor class. An additional variable that can be read from an eng file is the dry weight (the first line of an eng file specifies propellant mass and loaded/total mass [(source)](https://www.thrustcurve.org/info/raspformat.html), dry mass being the difference) for any motor class, and the initial mass for the GenericMotor class.

It could make RocketPy an easier tool to use to quickly compare how a rocket would perform using different motors, similar to [Thrust Curve's motor guide](https://www.thrustcurve.org/motors/guide.html) but with a more advanced flight simulator.

**Describe the solution you'd like**

1) The GenericMotor class gets the same update that the other motor classes did in #309 
2) All motor classes that have a thrust curve defined by an eng file have their dry masses defined by the eng file by default (a user-supplied dry mass would override that value)
3) The GenericMotor class has its propellant_initial_mass parameter defined by the eng file by default (a user-supplied propellant mass would override that value)

Gui-FernandesBR, 2024-05-16:
This is a good suggestion. @chasgior214 would you like to participate on the solution too?
We can guide you through the implementation.

chasgior214, 2024-05-16:
Sure! I've never used pull requests or contributed to a large project before, so apologies in advance if I mess something up lol. Where should I get started?

Gui-FernandesBR, 2024-07-24:
@juliomachad0 :

```python
from rocketpy.motors import GenericMotor

filepath = "path/to_your/file.eng"

# Case 1: simplest version
your_motor = GenericMotor.load_from_eng_file("path/to_your/file.eng")
your_motor.all_info()

# Case 2: overriding values
your_motor = GenericMotor.load_from_eng_file(
    "path/to_your/file.eng",
    dry_mass=10, # passing the dray_mass will override the mass written in the file
)
your_motor.all_info()
```

closed 2024-09-08 by Gui-FernandesBR