RocketPy-Team/RocketPy #606: RailButtons does not have rocket_radius attribute

djangovanderplas, 2024-05-20:
**Describe the bug**
When trying to add `RailButtons` using the `add_surfaces()` method, RocketPy crashes because it requires a rocket_radius for all surfaces being added (this is due to Flight.u_dot_generalized() method). This method works fine for fins, tail and nosecone but doesn't for rail buttons (I also suspect for the aerobrakes).

**To Reproduce**

I have included the standard rocketpy Calisto script, but I added the RailButtons using this method (for fins I can add them this way). RocketPy crashes when it is trying to plot the flight details.
```python
import rocketpy

env = rocketpy.Environment(latitude=32.990254, longitude=-106.974998, elevation=1400)
env.set_atmospheric_model(type="standard_atmosphere")

Pro75M1670 = rocketpy.SolidMotor(
    thrust_source=1000,
    dry_mass=1.815,
    dry_inertia=(0.125, 0.125, 0.002),
    nozzle_radius=33 / 1000,
    grain_number=5,
    grain_density=1815,
    grain_outer_radius=33 / 1000,
    grain_initial_inner_radius=15 / 1000,
    grain_initial_height=120 / 1000,
    grain_separation=5 / 1000,
    grains_center_of_mass_position=0.397,
    center_of_dry_mass_position=0.317,
    nozzle_position=0,
    burn_time=3.9,
    throat_radius=11 / 1000,
    coordinate_system_orientation="nozzle_to_combustion_chamber",
)

calisto = rocketpy.Rocket(
    radius=127 / 2000,
    mass=14.426,
    inertia=(6.321, 6.321, 0.034),
    power_off_drag=0.65,
    power_on_drag=0.65,
    center_of_mass_without_motor=0,
    coordinate_system_orientation="tail_to_nose",
)

calisto.add_motor(Pro75M1670, position=-1.255)

nose_cone = calisto.add_nose(
    length=0.55829, kind="von karman", position=1.278
)

fin_set = calisto.add_trapezoidal_fins(
    n=4,
    root_chord=0.120,
    tip_chord=0.060,
    span=0.110,
    position=-1.04956,
    cant_angle=0.5,
)

tail = calisto.add_tail(
    top_radius=0.0635, bottom_radius=0.0435, length=0.060, position=-1.194656
)

# This is the interesting part here
rail_buttons = rocketpy.RailButtons(buttons_distance=0.0818+0.6182)
rail_buttons_position = -0.6182
calisto.add_surfaces(rail_buttons, rail_buttons_position)

test_flight = rocketpy.Flight(
    rocket=calisto, environment=env, rail_length=5.2, inclination=85, heading=0
    )

test_flight.all_info()
```

**Expected behavior**

I would expect this to just work as it is the way to add AeroSurfaces Abstract Baseclasses to the rocket class. However, it crashes! I can solve this issue by stating:
```python
rail_buttons = rocketpy.RailButtons(buttons_distance=0.0818+0.6182)
rail_buttons_position = -0.6182
rail_buttons.rocket_radius = 127 / 2000 # Adding this line solves the problem
calisto.add_surfaces(rail_buttons, rail_buttons_position)
```
However, this might not be very intuitive to the user, as it is not required for other AeroSurfaces.

**Additional context**

This is the output for the script I provided:
```
capi_return is NULL
Call-back cb_f_in_lsoda__user__routines failed.
Fatal Python error: F2PySwapThreadLocalCallbackPtr: F2PySwapThreadLocalCallbackPtr: PyLong_AsVoidPtr failed
Python runtime state: initialized
Traceback (most recent call last):
  File "/Users/djangovanderplas/micromamba/envs/python312_env/lib/python3.12/site-packages/scipy/integrate/_ivp/base.py", line 154, in fun
    return self.fun_single(t, y)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/djangovanderplas/micromamba/envs/python312_env/lib/python3.12/site-packages/scipy/integrate/_ivp/base.py", line 23, in fun_wrapped
    return np.asarray(fun(t, y), dtype=dtype)
                      ^^^^^^^^^
  File "/Users/djangovanderplas/micromamba/envs/python312_env/lib/python3.12/site-packages/rocketpy/simulation/flight.py", line 1759, in u_dot_generalized
    surface_radius = aero_surface.rocket_radius
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'RailButtons' object has no attribute 'rocket_radius'
```

I am on RocketPy version 1.2.2

Gui-FernandesBR, 2024-05-20:
Thank you for reporting, @djangovanderplas !

The issue #561 is a related to this BUG.

As a pragmatical recommendation: don't use the `add_surfaces` to add the rail buttons. Instead, use the method `set_rail_buttons`, which is working fine.

Your issue is still valid, however, and we should work on a solution.

Gui-FernandesBR, 2024-07-20:
Solved by #643

Thank you for opening this issue @djangovanderplas !! We really appreciate it.

closed 2024-07-20 by Gui-FernandesBR