RocketPy-Team/RocketPy #625: _First Simulation with RocketPy_ Notebook File fails

rickhg12hs, 2024-06-25:
**Describe the bug**

A description of what the bug is.

`TypeError: 'numpy.float32' object is not iterable`

**To Reproduce**

Code and steps to reproduce the behavior.

Run `jupyter` notebook of the first simulation from the documentation and observe the error/failure.

**Expected behavior**

A description of what you expected to happen.

The first simulation should not fail.  It should showcase run without errors.

**Screenshots**

If applicable, add screenshots to help explain your problem.

```python
env.set_atmospheric_model(type="Forecast", file="GFS")

    oc_open: server error retrieving url: code=0 message="/gfs_0p25/gfs20240625/gfs_0p25_18z is not an available dataset"

We can see what the weather will look like by calling the info method!

env.max_expected_height = 5000  # adjust the plots to this height
env.info()

    
    Gravity Details
    
    Acceleration of gravity at surface level:    9.7911 m/s²
    Acceleration of gravity at   5.000 km (ASL): 9.7802 m/s²
    
    
    Launch Site Details
    
    Launch Date: 2024-06-26 12:00:00 UTC
    Launch Site Latitude: 32.99025°
    Launch Site Longitude: -106.97500°
    Reference Datum: SIRGAS2000
    Launch Site UTM coordinates: 315468.64 W    3651938.65 N
    Launch Site UTM zone: 13S
    Launch Site Surface Elevation: 1471.5 m
    
    
    Atmospheric Model Details
    
    Atmospheric Model Type: Forecast
    Forecast Maximum Height: 5.000 km
    Forecast Time Period: From  2024-06-25 12:00:00  to  2024-07-11 12:00:00  UTC
    Forecast Hour Interval: 3  hrs
    Forecast Latitude Range: From  -90.0 ° To  90.0 °
    Forecast Longitude Range: From  0.0 ° To  359.75 °
    
    
    Surface Atmospheric Conditions
    

    ---------------------------------------------------------------------------

    TypeError                                 Traceback (most recent call last)

    Cell In[7], line 2
          1 env.max_expected_height = 5000  # adjust the plots to this height
    ----> 2 env.info()

    File ~/ČVUT/B232/BE3MPVTY1-TeamWork/RocketPy/.venv/lib64/python3.12/site-packages/rocketpy/environment/environment.py:3234, in Environment.info(self)
       3225 def info(self):
       3226     """Prints most important data and graphs available about the
       3227     Environment.
       3228 
       (...)
       3231     None
       3232     """
    -> 3234     self.prints.all()
       3235     self.plots.info()
       3236     return None

    File ~/ČVUT/B232/BE3MPVTY1-TeamWork/RocketPy/.venv/lib64/python3.12/site-packages/rocketpy/prints/environment_prints.py:219, in _EnvironmentPrints.all(self)
        216 print()
        218 # Print atmospheric conditions
    --> 219 self.atmospheric_conditions()
        220 print()
        222 self.print_earth_details()

    File ~/ČVUT/B232/BE3MPVTY1-TeamWork/RocketPy/.venv/lib64/python3.12/site-packages/rocketpy/prints/environment_prints.py:144, in _EnvironmentPrints.atmospheric_conditions(self)
        135 """Prints atmospheric conditions.
        136 
        137 Returns
        138 -------
        139 None
        140 """
        141 print("\nSurface Atmospheric Conditions\n")
        142 print(
        143     "Surface Wind Speed: {:.2f} m/s".format(
    --> 144         self.environment.wind_speed(self.environment.elevation)
        145     )
        146 )
        147 print(
        148     "Surface Wind Direction: {:.2f}°".format(
        149         self.environment.wind_direction(self.environment.elevation)
        150     )
        151 )
        152 print(
        153     "Surface Wind Heading: {:.2f}°".format(
        154         self.environment.wind_heading(self.environment.elevation)
        155     )
        156 )

    File ~/ČVUT/B232/BE3MPVTY1-TeamWork/RocketPy/.venv/lib64/python3.12/site-packages/rocketpy/mathutils/function.py:1099, in Function.__call__(self, *args)
       1097     return self.plot()
       1098 else:
    -> 1099     return self.get_value(*args)

    File ~/ČVUT/B232/BE3MPVTY1-TeamWork/RocketPy/.venv/lib64/python3.12/site-packages/rocketpy/mathutils/function.py:886, in Function.get_value(self, *args)
        883     if isinstance(args[0], (int, float, complex, np.integer)):
        884         args = [list(args)]
    --> 886 x = list(args[0])
        887 x = list(map(self.get_value_opt, x))
        888 if isinstance(args[0], np.ndarray):

    TypeError: 'numpy.float32' object is not iterable
```

**Additional context**

Add any other context about the problem here.

```bash
(.venv) $ python --version
Python 3.12.3
(.venv) $ python -m pip list
Package                   Version
------------------------- --------------
anyio                     4.4.0
argon2-cffi               23.1.0
argon2-cffi-bindings      21.2.0
arrow                     1.3.0
asttokens                 2.4.1
async-lru                 2.0.4
attrs                     23.2.0
Babel                     2.15.0
beautifulsoup4            4.12.3
black                     24.4.2
bleach                    6.1.0
bs4                       0.0.2
certifi                   2024.6.2
cffi                      1.16.0
cftime                    1.6.4
charset-normalizer        3.3.2
click                     8.1.7
comm                      0.2.2
contourpy                 1.2.1
cycler                    0.12.1
debugpy                   1.8.2
decorator                 5.1.1
defusedxml                0.7.1
executing                 2.0.1
fastjsonschema            2.20.0
fonttools                 4.53.0
fqdn                      1.5.1
h11                       0.14.0
h3                        3.7.7
httpcore                  1.0.5
httpx                     0.27.0
idna                      3.7
imageio                   2.34.2
ipykernel                 6.29.4
ipympl                    0.9.4
ipython                   8.25.0
ipython-genutils          0.2.0
ipywidgets                8.1.3
isoduration               20.11.0
jedi                      0.19.1
Jinja2                    3.1.4
JPype1                    1.5.0
json5                     0.9.25
jsonpickle                3.2.2
jsonpointer               3.0.0
jsonschema                4.22.0
jsonschema-specifications 2023.12.1
jupyter                   1.0.0
jupyter_client            8.6.2
jupyter-console           6.6.3
jupyter_core              5.7.2
jupyter-events            0.10.0
jupyter-lsp               2.2.5
jupyter_server            2.14.1
jupyter_server_terminals  0.5.3
jupyterlab                4.2.2
jupyterlab_pygments       0.3.0
jupyterlab_server         2.27.2
jupyterlab_widgets        3.0.11
kiwisolver                1.4.5
lxml                      5.2.2
MarkupSafe                2.1.5
matplotlib                3.9.0
matplotlib-inline         0.1.7
mistune                   3.0.2
mypy-extensions           1.0.0
nbclient                  0.10.0
nbconvert                 7.16.4
nbformat                  5.10.4
nest-asyncio              1.6.0
netCDF4                   1.7.1.post1
notebook                  7.2.1
notebook_shim             0.2.4
numpy                     2.0.0
orhelper                  0.1.3
overrides                 7.7.0
packaging                 24.1
pandocfilters             1.5.1
parso                     0.8.4
pathspec                  0.12.1
pexpect                   4.9.0
pillow                    10.3.0
pip                       24.1
platformdirs              4.2.2
prometheus_client         0.20.0
prompt_toolkit            3.0.47
psutil                    6.0.0
ptyprocess                0.7.0
pure-eval                 0.2.2
pycparser                 2.22
Pygments                  2.18.0
pyparsing                 3.1.2
python-dateutil           2.9.0.post0
python-json-logger        2.0.7
pytz                      2024.1
PyYAML                    6.0.1
pyzmq                     26.0.3
qtconsole                 5.5.2
QtPy                      2.4.1
referencing               0.35.1
requests                  2.32.3
rfc3339-validator         0.1.4
rfc3986-validator         0.1.1
rocketpy                  1.3.0.post1
rocketserializer          0.2.0
rpds-py                   0.18.1
scipy                     1.14.0
Send2Trash                1.8.3
setuptools                70.1.1
simplekml                 1.3.6
six                       1.16.0
sniffio                   1.3.1
soupsieve                 2.5
stack-data                0.6.3
terminado                 0.18.1
timezonefinder            6.5.2
tinycss2                  1.3.0
tokenize-rt               5.2.0
tornado                   6.4.1
traitlets                 5.14.3
types-python-dateutil     2.9.0.20240316
uri-template              1.3.0
urllib3                   2.2.2
wcwidth                   0.2.13
webcolors                 24.6.0
webencodings              0.5.1
websocket-client          1.8.0
widgetsnbextension        4.0.11
windrose                  1.9.1
```

Gui-FernandesBR, 2024-06-25:
Thanks for the Bug report. We will investigate it ASAP

Gui-FernandesBR, 2024-06-25:
I recommend you downgrade the numpy version. Run `pip install numpy==1.26`

Numpy decided to have their "first big release since 2006", and now some libs are facing breaking changes, we included.

The corrections for numpy 2.0.0 are coming soon. Meanwhile please downgrade.

rickhg12hs, 2024-06-25:
Well after downgrading `numpy`, the notebook survives `env.info()`, but the following cell crashes.  8-(
```python
Pro75M1670 = SolidMotor(
    thrust_source="../../data/motors/Cesaroni_M1670.eng",
    dry_mass=1.815,
    dry_inertia=(0.125, 0.125, 0.002),
    nozzle_radius=33 / 1000,
    grain_number=5,
    grain_density=1815,
    grain_outer_radius=33 / 1000,
    grain_initial_inner_radius=15 / 1000,
    grain_initial_height=120 / 1000,
    grain_separation=5 / 1000,
    grains_center_of_mass_position=0.397,
    center_of_dry_mass_position=0.317,
    nozzle_position=0,
    burn_time=3.9,
    throat_radius=11 / 1000,
    coordinate_system_orientation="nozzle_to_combustion_chamber",
)

    ---------------------------------------------------------------------------

    FileNotFoundError                         Traceback (most recent call last)

    Cell In[8], line 1
    ----> 1 Pro75M1670 = SolidMotor(
          2     thrust_source="../../data/motors/Cesaroni_M1670.eng",
          3     dry_mass=1.815,
          4     dry_inertia=(0.125, 0.125, 0.002),
          5     nozzle_radius=33 / 1000,
          6     grain_number=5,
          7     grain_density=1815,
          8     grain_outer_radius=33 / 1000,
          9     grain_initial_inner_radius=15 / 1000,
         10     grain_initial_height=120 / 1000,
         11     grain_separation=5 / 1000,
         12     grains_center_of_mass_position=0.397,
         13     center_of_dry_mass_position=0.317,
         14     nozzle_position=0,
         15     burn_time=3.9,
         16     throat_radius=11 / 1000,
         17     coordinate_system_orientation="nozzle_to_combustion_chamber",
         18 )

    File ~/ČVUT/B232/BE3MPVTY1-TeamWork/RocketPy/.venv/lib64/python3.12/site-packages/rocketpy/motors/solid_motor.py:304, in SolidMotor.__init__(self, thrust_source, dry_mass, dry_inertia, nozzle_radius, grain_number, grain_density, grain_outer_radius, grain_initial_inner_radius, grain_initial_height, grain_separation, grains_center_of_mass_position, center_of_dry_mass_position, nozzle_position, burn_time, throat_radius, reshape_thrust_curve, interpolation_method, coordinate_system_orientation)
        184 def __init__(
        185     self,
        186     thrust_source,
       (...)
        203     coordinate_system_orientation="nozzle_to_combustion_chamber",
        204 ):
        205     """Initialize Motor class, process thrust curve and geometrical
        206     parameters and store results.
        207 
       (...)
        302     None
        303     """
    --> 304     super().__init__(
        305         thrust_source,
        306         dry_mass,
        307         dry_inertia,
        308         nozzle_radius,
        309         center_of_dry_mass_position,
        310         nozzle_position,
        311         burn_time,
        312         reshape_thrust_curve,
        313         interpolation_method,
        314         coordinate_system_orientation,
        315     )
        316     # Nozzle parameters
        317     self.throat_radius = throat_radius

    File ~/ČVUT/B232/BE3MPVTY1-TeamWork/RocketPy/.venv/lib64/python3.12/site-packages/rocketpy/motors/motor.py:272, in Motor.__init__(self, thrust_source, dry_mass, dry_inertia, nozzle_radius, center_of_dry_mass_position, nozzle_position, burn_time, reshape_thrust_curve, interpolation_method, coordinate_system_orientation)
        270 if isinstance(thrust_source, str):
        271     if thrust_source[-3:] == "eng":
    --> 272         _, _, points = Motor.import_eng(thrust_source)
        273         thrust_source = points
        275 # Evaluate raw thrust source

    File ~/ČVUT/B232/BE3MPVTY1-TeamWork/RocketPy/.venv/lib64/python3.12/site-packages/rocketpy/motors/mot
or.py:969, in Motor.import_eng(file_name)
        966 data_points = [[0, 0]]
        968 # Open and read .eng file
    --> 969 with open(file_name) as file:
        970     for line in file:
        971         if re.search(r";.*", line):
        972             # Extract comment

    FileNotFoundError: [Errno 2] No such file or directory: '../../data/motors/Cesaroni_M1670.eng'

```

Gui-FernandesBR, 2024-06-25:
> Well after downgrading `numpy`, the notebook survives `env.info()`, but the following cell crashes.  8-(
> ```python
> Pro75M1670 = SolidMotor( [...]

Well now it's a `FileNotFoundError`

Please verify whether you have the  '../../data/motors/Cesaroni_M1670.eng' path correctly set.

rickhg12hs, 2024-06-25:
I'm sure I don't.

What exactly is necessary to successfully run _First Simulation with RocketPy_?

Gui-FernandesBR, 2024-06-25:
Please navigate through this example:

https://colab.research.google.com/github/RocketPy-Team/rocketpy/blob/master/docs/notebooks/getting_started_colab.ipynb

lucasfourier, 2024-06-30:
@rickhg12hs Were you able to run the simulation?

rickhg12hs, 2024-06-30:
The `colab... ` works fine.

Gui-FernandesBR, 2024-08-03:
I fixed the broken documentation link with this PR: #645 

Please follow that PR to track this issue solving.

closed 2024-08-04 by Gui-FernandesBR