RocketPy-Team/RocketPy #650: Google Colab link on "First Simulation" doc page doesn't seem to work

jbliesener, 2024-08-05:
**Describe the bug**

"Google Colab" button on "First Documentation" page doesn't work.

**To Reproduce**

- Open doc on [First Simulation](https://docs.rocketpy.org/en/latest/user/first_simulation.html)
- Click on "Google Colab" in the "See also" box
- Google Colab opens, but can't find the notebook: 
```
Could not find getting_started_colab.html in https://api.github.com/repos/RocketPy-Team/rocketpy/contents/docs/notebooks?per_page=100&ref=master
CustomError: Could not find getting_started_colab.html in https://api.github.com/repos/RocketPy-Team/rocketpy/contents/docs/notebooks?per_page=100&ref=master
    at new FO (https://ssl.gstatic.com/colaboratory-static/common/9aff04dde2358fc7be6d8d16cc2c62e5/external_binary_l10n__de.js:2450:68)
    at wa.program_ (https://ssl.gstatic.com/colaboratory-static/common/9aff04dde2358fc7be6d8d16cc2c62e5/external_binary_l10n__de.js:2437:442)
    at za (https://ssl.gstatic.com/colaboratory-static/common/9aff04dde2358fc7be6d8d16cc2c62e5/external_binary_l10n__de.js:16:57)
    at wa.next_ (https://ssl.gstatic.com/colaboratory-static/common/9aff04dde2358fc7be6d8d16cc2c62e5/external_binary_l10n__de.js:14:195)
    at kaa.next (https://ssl.gstatic.com/colaboratory-static/common/9aff04dde2358fc7be6d8d16cc2c62e5/external_binary_l10n__de.js:17:48)
    at b (https://ssl.gstatic.com/colaboratory-static/common/9aff04dde2358fc7be6d8d16cc2c62e5/external_binary_l10n__de.js:17:309)
```

**Expected behavior**

Well, I'd expect something like a Jupyter notebook.

**Screenshots**

Well, it's in German...
[image]

Gui-FernandesBR, 2024-08-05:
@jbliesener the correct link is https://colab.research.google.com/github/RocketPy-Team/rocketpy/blob/master/docs/notebooks/getting_started_colab.ipynb, 

I already fixed it but I'm still waiting for the solution to get to the get to production.

Something is wrong with our documentation building process right now. I will fix it asap.

[image]

Thanks for opening the issue!

jbliesener, 2024-08-05:
Thanks for making RocketPy!

closed 2024-10-11 by Gui-FernandesBR