RocketPy-Team/RocketPy #655: ENH: 3-DOF simulation

Gui-FernandesBR, 2024-08-19:
Allow RocketPy to run a point-mass simulation (3DOF)

Gui-FernandesBR, 2024-09-13:
Could help us on this issue: https://github.com/werocketry/RocketFlightSim/tree/main/rocketflightsim

@chasgior214

chasgior214, 2024-09-15:
For sure! 

What is the goal of having 3 DOF simulations in RocketPy? I think the best way to implement it would depend on the goal(s) of having it.

LUCKIN13, 2024-11-17:
The ability to choose between 6DoF or 3DoF simulation modes allows users to simulate without requiring extensive data, such as the rocket's inertia. This makes RocketPy useful for preliminary design phases, where only basic parameters like mass and drag are needed to estimate the rocket's apogee, without concern for stability.

This feature will need to undergo testing, but we believe it could become a valuable tool, enabling RocketPy to support the rocket's development process from initial design to launch.