RocketPy-Team/RocketPy #699: Callback function for collecting additional data from Monte Carlo sims

emtee14, 2024-10-01:
**Is your feature request related to a problem? Please describe.**

When running Monte Carlo sims currently you are limited to the predefined variables when exporting the results. For certain applications, I'd like to be able to get other data points that aren't already in the default list or generate my own variables from the Flight class after the sim has run and add them to the exported dict. My primary use for this would be to export environmental data as at the moment when aggregating multiple sims together with multiple different dates/times or weather forecasts the only way to differentiate is by embedding the data in the file name which becomes quite arduous when you have a lot of additional data you want to add. It can also be useful to export sensor data since it can then be used for sensor mocking.

**Describe the solution you'd like**

My idea to fix this would be a function you pass to the MonteCarlo class which takes a Flight as an argument and returns a dict which can then be appended to the output dict of the simulation. This would allow other data points to be exported like environmental data and sensor data from the rocket.

**Additional context**

emtee14, 2024-10-01:
Also to add myself or someone from LURA is happy to implement this

Gui-FernandesBR, 2024-11-02:
related: #269

Gui-FernandesBR, 2024-11-23:
This is coming as a new feature in the next release!

closed 2024-11-23 by Gui-FernandesBR