RocketPy-Team/RocketPy #705: Unable to run env_gfs.set_atmospheric_model(type="forecast", file="GFS") | Please Help!!!

whitethunder007, 2024-10-08:
I'm trying to follow along with the first simulation example and when I try to set my environment I get a huge error about fetch function and netCDF3, please see the image for more context. How do I fix this, I have no clue!

[image]

Gui-FernandesBR, 2024-10-24:
@whitethunder007 this behavior is a bit odd to me since we do not use netcdf3 on rocketpy at all. Actually, you would want to use netcdf4 instead! It is possible that the GFS server provided a dataset during some unstable period.

Could you try running the code again and verify if the errors still stands?

Id so, can you tell us which python and rocketpy version are you using?

Gui-FernandesBR, 2024-11-02:
I just went trhough the google colab basic example notebook and could not reproduce the error.
I'm marking the issue as resolved.

@whitethunder007 , please feel free to re-open the thread in case you face the same problem again.

closed 2024-11-02 by Gui-FernandesBR