RocketPy-Team/RocketPy #709: TST: Increase code coverage to 90%

Gui-FernandesBR, 2024-10-09:
For a python library like ours, a 90% coverage threshold should be the ideal.
It should already consider the slow tests as well.

Coverage is not the only metric that matters, but it is the best one.

Gui-FernandesBR, 2024-11-02:
As of today, we have a cc of 87% when considering the slow tests. 
[image]

Unfortunately, when we exclude the slow tests this number reduced to only 76%.

**Ideally, given that RocketPy is a library, code coverage should be of 99.9% with slow tests and 90% without the slow tests**

ting-hong-shieh, 2026-08-16:
Yes. I re-measured first, because the baseline in my earlier comment was taken at `cb6106a717207dd8fc2dfe1446d80ff75022f21b` and 30 pull requests have landed since.

## Current baseline

At `4263fa95d7fe6f63d9593f01e4ff7a088369e195`, running the five steps of `test_pytest.yaml` in order (unit, doctests, integration with the three animation deselects, the VTK animation tests, acceptance), each `--cov-append`:

| Metric | Result |
| --- | ---: |
| Statements | 17,527 |
| Covered | 14,771 |
| Missed | 2,756 |
| Coverage | 84.2757% |

That is 84.28% without slow tests, up from the 80.85% I measured on 14 August. To reach 90% from here, 1,004 more statements have to be covered.

The VTK animation tests passed locally this time (3 passed in 5.93s), so `plots/flight_plots.py` measures 93.7% rather than the figure in my earlier table. The gap I reported there was an artifact of excluding those tests after the segmentation fault, not a real hole.

## Where the remaining 2,756 missed statements are

| Module | Statements | Missed | Coverage |
| --- | ---: | ---: | ---: |
| `plots/environment_analysis_plots.py` | 556 | 517 | 7.0% |
| `environment/environment_analysis.py` | 688 | 415 | 39.7% |
| `simulation/monte_carlo.py` | 542 | 179 | 67.0% |
| `mathutils/function.py` | 1,234 | 157 | 87.3% |
| `environment/environment.py` | 1,173 | 150 | 87.2% |
| `plots/monte_carlo_plots.py` | 241 | 89 | 63.1% |
| `plots/flight_plots.py` | 1,197 | 76 | 93.7% |
| `plots/rocket_plots.py` | 291 | 76 | 73.9% |
| `prints/environment_analysis_prints.py` | 67 | 54 | 19.4% |

The first two and the ninth are one subsystem, and together they account for 986 of the 1,004 statements needed. `EnvironmentAnalysis` is not untested: it has ten tests, and nine of them carry `@pytest.mark.slow`, so the default suite never runs them.

Taking the union of the baseline run with those ten tests:

| Metric | Result |
| --- | ---: |
| Covered | 15,716 |
| Coverage | 89.6707% |

So the existing tests, if the default suite could run them, close all but 59 statements of the distance to 90%.

## Why they are marked slow, and what it costs

The fixture reads two committed files under `data/weather/`, so nothing here touches the network. It is `@pytest.fixture` at function scope, and nine tests request it, so twenty years of hourly reanalysis data is parsed nine times:

```
$ pytest tests/unit/environment/test_environment_analysis.py \
         tests/integration/environment/test_environment_analysis.py \
         --runslow -q --durations=15
25.43s setup    tests/unit/environment/test_environment_analysis.py::test_values
24.20s setup    tests/unit/environment/test_environment_analysis.py::test_profile_plots
23.58s setup    tests/integration/environment/test_environment_analysis.py::test_create_environment_object
23.54s setup    tests/unit/environment/test_environment_analysis.py::test_pressure_level_wind_profile_uses_velocity_components
23.53s setup    tests/unit/environment/test_environment_analysis.py::test_animation_plots
23.43s setup    tests/integration/environment/test_environment_analysis.py::test_all_info
23.37s setup    tests/unit/environment/test_environment_analysis.py::test_distribution_plots
23.35s setup    tests/unit/environment/test_environment_analysis.py::test_average_plots
23.34s setup    tests/integration/environment/test_environment_analysis.py::test_exports
14.69s call     tests/integration/environment/test_environment_analysis.py::test_exports
 8.72s call     tests/integration/environment/test_environment_analysis.py::test_all_info
 6.52s call     tests/unit/environment/test_environment_analysis.py::test_animation_plots
10 passed in 246.92s
```

Roughly 212 of those 247 seconds are the same fixture being rebuilt. Setting `scope="session"` on it takes the same ten tests to 30.11s. Measuring both with a warm page cache so the comparison is fair, it is 106.70s before and 30.11s after.

No test mutates the fixture; `test_exports` already deep-copies it before calling `load`. At 30s for the whole file, the nine `slow` marks no longer buy anything, and the tests can move into the default suite.

## Measured with that change

I scoped the fixture to the session, dropped the nine marks, and re-ran the same five steps:

| Metric | Before | After |
| --- | ---: | ---: |
| Covered | 14,771 | 15,716 |
| Missed | 2,756 | 1,811 |
| Coverage | 84.2757% | 89.6674% |

| Module | Before | After |
| --- | ---: | ---: |
| `plots/environment_analysis_plots.py` | 7.0% | 98.4% |
| `environment/environment_analysis.py` | 39.7% | 94.2% |
| `prints/environment_analysis_prints.py` | 19.4% | 100.0% |

Nothing failed: unit 2,173 passed and 7 skipped, doctests 48 passed, integration 157 passed and 37 skipped, the three animation tests passed, acceptance 18 passed. The cost is about 74 seconds across the five steps, and about 84 MB of resident memory held for the session.

That is #1176, test-only. Twenty slow marks remain, twelve in `tests/integration/environment/test_environment.py` and the rest across the Monte Carlo, flight, encoding and Open-Meteo tests; those are slow for reasons a fixture scope does not address, and `test-pytest-slow.yaml` still runs them.

That leaves 59 statements to 90%, which is one ordinary slice. `plots/rocket_plots.py` at 73.9% and `plots/monte_carlo_plots.py` at 63.1% are the next two that are deterministic and need no network, and I would take them the way I took #1152 and #1165.

Measured on Python 3.12.3, Linux, `MPLBACKEND=Agg`, with the `[all]` extras installed.

ting-hong-shieh, 2026-08-16:
Correcting the absolute figures in my previous comment. The measurement environment was missing an optional dependency, so the numbers were low. The delta was right; the baseline was not.

## What was wrong

`contextily` is declared in the `monte-carlo` extra in `pyproject.toml`, but it is not in `requirements-optional.txt`, and `make install` installs from the requirements files:

```make
install:
	$(PYTHON) -m pip install --upgrade pip
	pip install -r requirements.txt
	pip install -r requirements-optional.txt
	pip install -r requirements-tests.txt
	pip install -e .
```

`test_monte_carlo_plots_background.py` opens with `pytest.importorskip("contextily")`, so the whole file is skipped. That is 18 tests and 53 statements, mostly in `_get_environment_coordinates`, `_resolve_map_provider` and `_get_background_map`. The workflow installs `.[all]`, which does pull `contextily` on Python 3.10, so CI runs them and I did not.

A contributor following `make install` gets a pass with 18 silent skips and a coverage number that does not match what Codecov reports for the same commit, with nothing pointing at the cause. That seems worth fixing separately; adding the same specifier to `requirements-optional.txt` is enough.

## The corrected figures

Installing `contextily` and re-running the same five steps at `4263fa9` gives a baseline that agrees with Codecov to the statement:

| | Re-measured | Codecov on #1176 | What I reported |
| --- | ---: | ---: | ---: |
| Covered | 14,824 | 14,824 | 14,771 |
| Missed | 2,703 | 2,703 | 2,756 |
| Coverage | 84.5781% | 84.57% | 84.2757% |

The 53-statement difference is that skipped file. `plots/monte_carlo_plots.py` alone moves from 89 missed to 36 once the background tests can run.

Codecov puts #1176 at 89.96%, and its `+945` matches what I measured, so the change itself was reported correctly. What was wrong is the distance still to go: from #1176 the 90% target is **6 statements away, not the 59 I reported**.

## What that changes

Nothing about #1176, and nothing about the finding that `EnvironmentAnalysis` was the subsystem holding the number down. It does mean the remaining work is much smaller than I said.

#1178 is the next slice, test-only: `plots/rocket_plots.py` from 73.8832% to 99.3127% and `plots/monte_carlo_plots.py` from 85.0622% to 96.6805%, +102 statements. Those two were uncovered because no fixture reaches the branches — a single `Fin` rather than a fin set, a `GenericSurface`, hybrid and liquid motors, a ring cluster, a nozzle sitting behind the last surface in each coordinate system, and the ellipse paths that run when a results file is missing its apogee or impact series.

#1176 and #1178 touch disjoint modules, so on the corrected baseline the pair clears 90%. I would rather let Codecov say so on each pull request than quote my own arithmetic again.