RocketPy-Team/RocketPy #716: TST: 2-stage Acceptance Teat

chasgior214, 2024-10-23:
**Is your feature request related to a problem? Please describe.**

With RocketPy getting more into multi-stage rockets, an acceptance test to verify the reliability of multi-stage simulations would be a good next step. 

**Describe the solution you'd like**

NASA has published a series of verification check-cases for atmospheric flight, available at https://nescacademy.nasa.gov/flightsim

Check-case 17, Two-stage rocket to orbit, would be a great addition to RocketPy's tests, and its first test case for a multi-stage rocket. While the initial implementation doesn't have to test against the whole flight, it could give verification to RocketPy's ability to simulate multi-stage rockets in the lower layers of the atmosphere, and in the future provide a way to verify orbital simulations (#277). 

**Additional context**

As noted in the abstract of the linked webpage:
> The rise of innovative unmanned aeronautical systems and the emergence of commercial space activities have resulted in a number of relatively new aerospace organizations that are designing innovative systems and solutions. These organizations use a variety of commercial off-the-shelf and in-house-developed simulation and analysis tools including 6-degree-of-freedom (6-DOF) flight simulation tools. The increased affordability of computing capability has made high-fidelity flight simulation practical for all participants. Verification of the tools' equations-of-motion and environment models (e.g., atmosphere, gravitation, and geodesy) is desirable to assure accuracy of results. However, aside from simple textbook examples, minimal verification data exists in open literature for 6-DOF flight simulation problems.

Adding these to RocketPy (with the two-stage rocket as the first step) would help bring it closer to industry-grade robustness and accuracy.

Gui-FernandesBR, 2024-10-24:
Related issue: #45 

-------------

@chasgior214 thank you so much for opening this issue. We were planning to improve the multi-stage support on rocketpy this year but we faced a few delays and decided to prioritized other implementations. Your suggested links may help us once we decide to get back to work on this feature, thx

Gui-FernandesBR, 2024-11-02:
Related: #662 and #45