RocketPy-Team/RocketPy #717: Problem with heading parameter definition

Inazillo, 2024-10-24:
I'm trying to simulate how the cross wind affects to the trajectory, this way I want to analyze in the extreme cases if the rocket will go out of the safe zone. For that I'm defining a wind_u of 30 m/s and wind_v of 0 m/s, moreover, the heading (angle relative to the north) of 11.6°.
The problem comes when I try to simulate it, as you can see on the attached figure that rocketpy is launching towards the west, instead of doing that towards the north with the defined angle. You can find a simplified code attached, so that you can review it and see if it's a bug or my error.
Thank you very much in advance!
[image]
[Airfoil.CSV](https://github.com/user-attachments/files/17504062/Airfoil.CSV)
[CD_power_off.CSV](https://github.com/user-attachments/files/17504063/CD_power_off.CSV)
[CD_power_on.CSV](https://github.com/user-attachments/files/17504064/CD_power_on.CSV)
[Wind_cross_flight.txt](https://github.com/user-attachments/files/17504099/Wind_cross_flight.txt)

Gui-FernandesBR, 2024-10-24:
There's something wrong with the wind_u definition, for sure. 
But it seems that this behavior has been present since v1.0 or even more.

So I have to check it deeper to properly understand it. Let me see what I can get during the next weeks.

Gui-FernandesBR, 2024-11-23:
Doing some investigations today. I think the problem might be related to the Flight.u_dotxxx methods @MateusStano