RocketPy-Team/RocketPy #718: Problem with NetCFD4 file from CDS

robbestephan, 2024-10-30:
**Describe the bug**

I run into problems with my NetCFD4 file gathered from CDS.

**To Reproduce**
This is my code, the .nc file is not supported to add as an attachment here.
I gathered the data from here: https://cds.climate.copernicus.eu/datasets/reanalysis-era5-pressure-levels?tab=overview

```
from rocketpy import Environment
import os
dirname = os.path.dirname(__file__)

Env = Environment(
    date=(2024, 10, 21, 17),
    latitude=50.483618,
    longitude=6.181357,
    elevation=559,
)

filename = os.path.join(dirname, 'Elsenborn_pressure_levels_reanalysis_21oct24_Reanalysis.nc'),

Env.set_atmospheric_model(
    type='Reanalysis',
    file=filename,
    dictionary="ECMWF",
)

Env.plots.atmospheric_model()
```

**Expected behavior**

I expected to get plots of the atmosphere and eventually implement it into our simulations

**Screenshots**

Here is the data request I did from CDS:
[image]

**Additional context**

Add any other context about the problem here.

robbestephan, 2024-11-15:
It seems that the NetCDF4 file I get from CDS have dict keys 'valid_time' instead of 'time' and 'pressure_level' instead of 'level' renaming them seems to have solved the problem

```
import xarray as xr

# Load the NetCDF file
file_path = "reanalysis.nc"
ds = xr.open_dataset(file_path)

# Rename time and pressure
ds = ds.rename({"valid_time": "time"})
ds = ds.rename({"pressure_level": "level"})

# Save the modified dataset
ds.to_netcdf("reanalysis_fix.nc")
```

Gui-FernandesBR, 2024-11-23:
Amazing work at #736 , I'm closing this issue as solved! NExt release (v1.7.0) should be out soon!

closed 2024-11-23 by Gui-FernandesBR