RocketPy-Team/RocketPy #723: DOC:

aureliobarbosa, 2024-11-07:
**Describe the bug**

The GitHub badge appearing on the right top badge menu of the documentation (just on the side of the Light/Dark, Linkedin, and PyPI badges) is duplicated.

**To Reproduce**

This 'bug' is appearing on the latest documentation documentation page, which at this  time corresponds to version 1.6.1.

**Expected behavior**

Just one badge should be present.

**Screenshots**
[image]

**Additional context**

Do not apply.

Gui-FernandesBR, 2024-11-07:
Thank you for opening this issue.
We are aware of such problem and the next minor version should already fix that, we plan to release it later this week.
#722

aureliobarbosa, 2024-11-07:
Thanks for answering quickly, I was planning to fix it. Shouldn't mark as closed or mention it on the PR that closes the issue?

Gui-FernandesBR, 2024-11-07:
> Thanks for answering quickly, I was planning to fix it. Shouldn't it be marked as closed or be mentioned on the PR that closes the issue?

Thanks for helping.
I will close this issue as soon as we merge #722 onto master.

closed 2024-11-07 by Gui-FernandesBR