RocketPy-Team/RocketPy #767: Add Angle of Attack parameter to power-on and power-off drag callable

NathanTPerkins, 2025-02-10:
**Is your feature request related to a problem? Please describe.**

Our rocketry club wants to use the power-on and power-off drag callable parameters, but right now the only input is free-stream-mach. We hope support could be added to include the current angle of attack of the rocket as another input parameter.

**Describe the solution you'd like**

We hope there could be another option for a parameter for the power-on and power-off drag callable parameter. 

**Additional context**

Gui-FernandesBR, 2025-02-10:
@NathanTPerkins , in RocketPy, the drag force is considered as the force that is parallel to the rocket's axis, regardless of the angle of attack

NathanTPerkins, 2025-02-12:
So our club is trying to do CFD on our rocket and we want to make Cd and function of angle of attack and Mach number. When rocketpy was being developed did you guys originally try to do this and then hit a dead end so you ended up with the current solution?

Gui-FernandesBR, 2025-02-12:
> When rocketpy was being developed did you guys originally try to do this and then hit a dead end so you ended up with the current solution?

No we didn't plan to do that.

It is not hard to do that in terms of coding, if you want to. Just follow the example of the lift coefficient derivative in the flight class.

I recommend reading RocketPy's paper in order to understand the flight modelling: https://doi.org/10.1061/(ASCE)AS.1943-5525.0001331

Gui-FernandesBR, 2026-11-02:
Reopened: the fix is merged on `develop` and will close when v1.14.0 is released to `master`.