RocketPy-Team/RocketPy #769: ENH: Add pressure thrust

caioessouza, 2025-02-11:
**Is your feature request related to a problem? Please describe.**

A description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A description of what you want to happen.

**Additional context**

Add any other context or screenshots about the feature request here.

- [ ] Review theory
- [ ] https://www.rocketryforum.com/threads/rasaero-ii-apogee-discrepancy-with-openrocket.170567/
- [ ] Check which parameter to add to the init of all motor classes
- [ ] Implemet a function like that "f = Function(lambda t,p: thrust(t) + (exit_pressure(t) - p)*Ae/(A_throat*combustion_pressure(t)))" into the motor class
- [ ] Impelement ambient pressure import from flight simulation u_dot functions

caioessouza, 2025-02-11:
Based on #148

kevin-alcaniz, 2025-03-19:
At Faraday, we've made some changes to account for pressure thrust and correct the thrust force as the rocket ascends. Would you like me to open a PR?

caioessouza, 2025-03-21:
> At Faraday, we've made some changes to account for pressure thrust and correct the thrust force as the rocket ascends. Would you like me to open a PR?

Sure, I just saw what you made at the issue #789 and it seems very good. Feel free to open the PR, thank you for the contribuition!

closed 2025-04-16 by Gui-FernandesBR