RocketPy-Team/RocketPy #772: Can't access Flight.x, Flight.y, Flight.z values after flight.

finnmezz, 2025-02-18:
Hi! The bug I'm experiencing is I am trying to access the values of Flight.x, Flight.y and Flight.z. I checked the documentation and Flight.x and Flight.z are indexed differently to Flight.y. I want to be able to index the x,y, and z coordinates of the rocket at a certain point in time (17.1 seconds). This is because then the second stage will launch. 

However, when I try to even use "display(type(Flight_stage1.x))" or anything similar, I get the error message:
`KeyError: 'x'`
`ValueError: array must not contain infs or NaNs`

I tried writing a script to turn infs and NaNs to zero but it seems to be a decorator format which I am not used to.

Here is the code I tried to run, which works up until  `display(type(Flight_stage1.y))`:
(You'll have to use a thrust curve file in your directory!)

````from rocketpy import Environment, Rocket, SolidMotor, Flight
import math
import datetime

Env1 = Environment(latitude=67.89325597913002, longitude=21.065756056273834, elevation=300) 
tomorrow = datetime.date.today() + datetime.timedelta(days=1)
Env1.set_date((tomorrow.year, tomorrow.month, tomorrow.day, 12))
Env1.set_atmospheric_model(type="Ensemble", file="GEFS")

Motor1 = SolidMotor(
    thrust_source=r"...Python Modelling\Thrust curves\Motor.eng",
    dry_mass=260, # kg (gross mass - propellant mass)
    dry_inertia=(261.89, 261.89, 10.98),
    nozzle_radius=0.23957143, # radius of the nozzle outlet in meteres
    grain_number=1,
    grain_density=1750,
    grain_outer_radius=0.2795,
    grain_initial_inner_radius=0.07985714,
    grain_initial_height=2.6072,
    grain_separation=0,
    grains_center_of_mass_position= 1.445, 
    center_of_dry_mass_position=0,
    nozzle_position=3.44,
    burn_time=13.1,
    throat_radius=0.07985714,
    coordinate_system_orientation="combustion_chamber_to_nozzle",
)

# Rocket

rocket_stage1 = Rocket(
    radius=0.483,
    mass=555.736, 
    inertia=(2688.81, 2688.81, 36.97), 
    power_off_drag=r"...\Python Modelling\Drag curves\Motor1PowerOffDragCurve.eng",
    power_on_drag=r"...\Python Modelling\Drag curves\Motor1PowerOnDragCurve.eng",
    center_of_mass_without_motor=3.88, # m
    coordinate_system_orientation="nose_to_tail",
)

rocket_stage1.add_motor(Motor1, position=4.893)

nose_cone = rocket_stage1.add_nose(length=2.49, kind="ogive", position=0)

fin_set_1 = rocket_stage1.add_trapezoidal_fins(
    n=4,
    root_chord=0.865,
    tip_chord=0.302,
    span=0.352,
    position=3.985,
    sweep_length = 0.707
)

# Defining triggers for parachutes
def drogueTrigger(p, y, x, z, already_triggered=False):
    if already_triggered:
        return False
    if y < 0:
        return True
    return False
    #return True if y < 0 else False  # Less than zero, so rocket is coming back down from apogee

def mainTrigger(p, y, x, z, already_triggered=False): # Extra arguments are given because trigger is passing a lot through
    if already_triggered:
        return False
    if y < 0 and y < 1000:
        return True
    return False
    #print(type(y))
    #print(y)
    #return True if y < 0 and y < 1000 else False # Parachute within 1km from ground

Main = rocket_stage1.add_parachute('Main',
                                   cd_s=10.0,
                                   trigger=mainTrigger, 
                                   sampling_rate=105,
                                   lag=1.5,
                                   noise=(0, 8.3, 0.5))

Drogue = rocket_stage1.add_parachute('Drogue',
                                   cd_s=1.0,
                                   trigger=drogueTrigger, 
                                   sampling_rate=105,
                                   lag=1.5,
                                   noise=(0, 8.3, 0.5))

                              # Parameters for first stage rocket flight
Flight_stage1 = Flight(rocket=rocket_stage1, environment=Env1, inclination=85, heading=0 ,verbose=True, rail_length=10)

ignition_delay = 4
tsecond_stage = RedKite.burn_duration + ignition_delay

display(type(Flight_stage1.y))
````

I would expect the type of Flight_stage1.y to be listed. I want to be able to call  `Flight.y(tsecond_stage)` for example, to see what the y value is at the time `t=tsecond_stage`.

I feel like this should be a simple fix! Any support on this I can get would be much appreciated.

Gui-FernandesBR, 2025-03-03:
Hello @finnmezz , thank you for your effort opening this issue and, of course, for your patience waiting for a response.

1. When creating a bug report, we apologize if you could provide the the motor .eng and rocket drag files, otherwise the code is unreproducible. I will ignore the files by overriding them with a constant value.
2. The `RedKite` variable is not defined, so the final codelines cannot be executed.
3. I could not clearly identify what exactly you were trying to do with your trigger functions. I guess there's a mistake in here when you write  `return True if if y < 0 and y < 1000 else False` (main trigger). I'm getting a lot of "WARNING: Trying to add flight phase starting *together* with the one *preceding* it.  This may be caused by multiple parachutes being triggered simultaneously." warnings. My suggestion is for you to change the triggers, try something like:
```python
Main = rocket_stage1.add_parachute('Main',
                                   cd_s=10.0,
                                   trigger=1000,  # 1km above ground level
                                   sampling_rate=105,
                                   lag=1.5,
                                   noise=(0, 8.3, 0.5))

Drogue = rocket_stage1.add_parachute('Drogue',
                                   cd_s=1.0,
                                   trigger="apogee",
                                   sampling_rate=105,
                                   lag=1.5,
                                   noise=(0, 8.3, 0.5))
```

4. Do you really mean your rocket weights `555.736` kg? 
5. I could obtain the simulation results, however they do not make a lot of sense, please use `rocket_stage1.all_info()` method to verify your model. I was surprised that the static margin was negative, which usually leads to a indefinite simulation. 
6. Last but not least, it's possible that you have problematic data on your csv files. Try to load and plot them one by one using the Function class in order to check if there's any missing or outlier value;

finnmezz, 2025-03-04:
Hi @Gui-FernandesBR , 
No worries, and thanks so much for getting back to me. This has pretty much fixed it for me. Thanks so much!

closed 2025-03-04 by finnmezz