RocketPy-Team/RocketPy #773: Bug Report: Runtime Error in RocketPy's add_motor Function

vkhansen, 2025-02-21:
Issue Summary:The program fails when executing the example_rocket.add_motor(hybrid_motor, position=0) command. This issue occurs even on previously working code, indicating a potential problem with internal calculations in the RocketPy library.

Steps to Reproduce:

Run a previously working RocketPy script that includes the following command:

example_rocket.add_motor(hybrid_motor, position=0)

The error occurs at runtime, preventing further execution.

Expected Behavior:

The motor should be added to the rocket without errors.

The static margin evaluation should proceed without encountering invalid numerical values.

Actual Behavior:

A RuntimeWarning is triggered due to division by zero and invalid scalar multiplications in mathutils/function.py:

C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\mathutils\function.py:2378:
RuntimeWarning: divide by zero encountered in scalar divide
return Function(lambda x: (self.get_value_opt(x) / other(x)))

C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\mathutils\function.py:2311:
RuntimeWarning: invalid value encountered in scalar multiply
return Function(lambda x: (self.get_value_opt(x) * other(x)), inputs)

This is followed by a traceback leading to a ValueError:

ValueError: array must not contain infs or NaNs

The error occurs while evaluating the static margin of the rocket, specifically within RocketPy’s interpolation and function evaluation methods.

```
Traceback:

Traceback (most recent call last):
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\6IN_OLD.py", line 149, in <module>
    example_rocket.add_motor(hybrid_motor, position=0)
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\rocket\rocket.py", line 991, in add_motor
    self.evaluate_static_margin()
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\rocket\rocket.py", line 655, in evaluate_static_margin
    self.static_margin.set_discrete(
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\mathutils\function.py", line 718, in set_discrete
    func.set_source(np.column_stack((xs, ys)))
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\mathutils\function.py", line 264, in set_source
    self.set_interpolation(self.__interpolation__)
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\mathutils\function.py", line 309, in set_interpolation
    self.__update_interpolation_coefficients(self.__interpolation__)
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\mathutils\function.py", line 324, in __update_interpolation_coefficients
    self.__interpolate_spline__()
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\rocketpy\mathutils\function.py", line 1907, in __interpolate_spline__
    c = linalg.solve_banded(
        ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\scipy\linalg\_basic.py", line 596, in solve_banded
    b1 = _asarray_validated(b, check_finite=check_finite, as_inexact=True)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\scipy\_lib\_util.py", line 537, in _asarray_validated
    a = toarray(a)
        ^^^^^^^^^^
  File "C:\Users\User\Documents\GitHub\rocket_trajectory\venv\Lib\site-packages\numpy\lib\_function_base_impl.py", line 646, in asarray_chkfinite
    raise ValueError(
ValueError: array must not contain infs or NaNs
```

[6IN_HYBRID.txt](https://github.com/user-attachments/files/18904127/6IN_HYBRID.txt)

Gui-FernandesBR, 2025-02-21:
This is the third time that I see someone reporting a bug related to "ValueError: array must not contain infs or NaNs"

I'm wondering whether there's a problem with the Function class or even numpy.

Thank you so much for reporting the bug, @vkhansen , specially for the level of details. We will investigate it and get back to you.

Gui-FernandesBR, 2025-03-03:
TLDR: try to comment this line: `burn_time=15,` or at least reduce the burn time to something between 14s and 15s.

---------------

@vkhansen I appreciate your bug report and all the details you have provided. 

The file that you have shared is what actually allowed me to navigate through your problem and try to debug it.
Fascinating how you organized your simulation, good one.

So it seems there's a problem with the static margin evaluation here.
I've plotted the static margin function  along with other interesting functions. It seems that rocket is not able to evaluate the S.M. when x gets above 14 seconds. 

[image]

[image]

[image]

```python
>>> for i in range(20):
           print(i, self.static_margin(i))

0 15.831374009427591
1 15.755324319452827
2 15.701218810353936
3 15.672508388801436
4 15.673172361510526
5 15.707802875678329
6 15.781726400019416
7 15.901133057137876
8 16.073213715388114
9 16.306391438525807
10 16.610182065914067
11 16.994312819563238
12 17.47296863991595
13 18.06431387165062
14 18.791526938520104
15 nan
16 nan
17 nan
18 nan
19 nan
15.831374009427591 nan
```

Therefore, the `Rocket.evaluate_static_margin` method is breaking at the final row, when it tries to discretize the static margin function.

I will add a clearer error message in order to help future users

Gui-FernandesBR, 2025-03-25:
So we to not add any particular error message to cover this specific problem. We understood this was a very particular case.

#780 will make sure the Rocket object does not crash until the static margin is actually accessed.

For now, I will mark this issue as closed.

closed 2025-03-25 by Gui-FernandesBR