RocketPy-Team/RocketPy #780: ENH: Make Static Margin Calculation Lazy

MateusStano, 2025-03-04:
Currently, the static margin calculation is performed during the simulation, which is unnecessary and potentially inefficient. This calculation should be deferred and performed only when it is explicitly needed. This adjustment could enhance the performance and streamline the simulation process.

Proposed Solution:

- Refactor the static margin calculation so that it is computed lazily.
- Ensure that the calculation is triggered only when the static margin value is specifically requested.
- Use some logic like that of `@cached_property`, so that it is only calculated when its need
- Make sure that whenever something is added/changed in the rocket that could chang the SM value (e.g. adding a new aero surface) the cached value is cleaned

Gui-FernandesBR, 2025-03-05:
This is not an easy request. The whole Rocket class is currently on top of "anticipated evaluations" rather than "lazy evaluations". 
It is important, of course, but not easy to be done.

Gui-FernandesBR, 2026-11-02:
Reopened: the fix is merged on `develop` and will close when v1.14.0 is released to `master`.