RocketPy-Team/RocketPy #782: Bug Report: Flight does not work for some "initial condition" inputs

AndyBankes, 2025-03-05:
**Issue:**

My script is attempting to eject a payload off of a High-Power Rocket. I have tried the following code as a test, which works for smaller rockets: [https://docs.rocketpy.org/en/latest/user/deployable.html](url)

However, the rocket I am simulating is much heavier and has a much higher thrust. I run the flight for the rocket so it stops at apogee and it can plot this flight, no problem. However, when I try to then "eject the payload" ("launch" another "rocket" with the initial conditions of the previous one), the loading time is indefinite, it tells me I have some NaN or inf values in the array, or it just plots a dot on a graph with no trajectory at 0,0,0.

**To Reproduce**

Copy-paste the .txt files into .eng files and save them into your directory:

[MOTOR.txt](https://github.com/user-attachments/files/19082260/MOTOR.txt)

[MOTOR_PowerOffDragCurve.txt](https://github.com/user-attachments/files/19082274/MOTOR_PowerOffDragCurve.txt)

[MOTOR_PowerOnDragCurve.txt](https://github.com/user-attachments/files/19082278/MOTOR_PowerOnDragCurve.txt)

[PAYLOAD_PowerOffDragCurve.txt](https://github.com/user-attachments/files/19082279/PAYLOAD_PowerOffDragCurve.txt)

[PAYLOAD_PowerOnDragCurve.txt](https://github.com/user-attachments/files/19082280/PAYLOAD_PowerOnDragCurve.txt)

Then attempt the script:

````
from rocketpy import Environment, SolidMotor, Rocket, Flight
import datetime

%matplotlib widget

env1 = Environment(latitude=-17.000000, longitude=140.000000, elevation=0) 
tomorrow = datetime.date.today() + datetime.timedelta(days=1)
env1.set_date((tomorrow.year, tomorrow.month, tomorrow.day, 12))
env1.set_atmospheric_model(type="Ensemble", file="GEFS")
````
````
motor1 = SolidMotor(
    thrust_source=r"...\MOTOR.eng",
    dry_mass=262, # kg 
    dry_inertia=(264.885, 264.885, 12.98), # 
    nozzle_radius=0.236, # 
    grain_number=1,
    grain_density=1750,
    grain_outer_radius=0.2590,
    grain_initial_inner_radius=0.0750,
    grain_initial_height=2.6082,
    grain_separation=0,
    grains_center_of_mass_position= 1.4681, 
    center_of_dry_mass_position=2.37,
    nozzle_position=3.44,
    burn_time=13.1,
    throat_radius=0.07985714,
    coordinate_system_orientation="combustion_chamber_to_nozzle",
)
print(f"Total Impulse of the Solid Motor: {motor1.total_impulse} Ns")
motor1.all_info()
````
````
rocket_stage_1 = Rocket(
    radius=0.3055, # widest radius of the rocket
    mass=843.239, # mass without motor and propellant in kg
    inertia=(2787.7, 2787.7, 61.18), # kgm^2, without motor or propellant mass
    power_off_drag=r"...\MOTOR_PowerOffDragCurve.eng",
    power_on_drag=r"...\MOTOR_PowerOnDragCurve.eng",
    center_of_mass_without_motor=3.51, # m
    coordinate_system_orientation="nose_to_tail",
)

nose_cone = rocket_stage_1.add_nose(length=3.19, kind="ogive", position=0, base_radius=0.3055)

rocket_stage_1.add_motor(motor1, position=4.483)

fin_set = rocket_stage_1.add_trapezoidal_fins(
    n=4,
    root_chord=0.703,
    tip_chord=0.703,
    span=0.547,
    position=7.25,
    sweep_length = 0.379
)
````
````
Flight_stage_1 = Flight(rocket=rocket_stage_1, environment=env1, rail_length=10.2, inclination=90, heading=0,terminate_on_apogee=True)
Flight_stage_1.plots.trajectory_3d()
print(Flight_stage_1.apogee)
print(Flight_stage_1.apogee_time)

Flight_stage_1.plots.trajectory_3d()
````
````
rocket_stage_2 = Rocket(
    radius=0.15, # widest radius of the payload
    mass=3.5, # kg
    inertia=(0.06924, 0.18838, 0.22481), # kgm^2, without motor or propellant mass
    power_off_drag=r"...\PAYLOAD_PowerOffDragCurve.eng",
    power_on_drag=r"...\PAYLOAD_PowerOnDragCurve.eng",
    center_of_mass_without_motor=0.503,
    coordinate_system_orientation="nose_to_tail",
)

nose_cone = rocket_stage_2.add_nose(length=0.27, kind="ogive", position=0, base_radius=0.15, power=1)

fin_set_2 = rocket_stage_2.add_trapezoidal_fins(
    n=2,
    root_chord=0.63905325443787,
    tip_chord=0.118343195266272,
    span=0.36,
    position=0.307692307692308,
    sweep_length = 0.42603550295858
)

transition_2 = rocket_stage_2.add_tail(
    top_radius= 0.15,
    bottom_radius = 0.1,
    length = 0.065,
    position = 0.93491124260355
)
````
````
Flight_stage_2 = Flight(
    rocket=rocket_stage_2, 
    environment=env2, 
    rail_length=1,
    inclination=Flight_stage_1.w1(tsecond_stage), 
    heading=0, 
    initial_solution = [tsecond_stage, Flight_stage_1.x(tsecond_stage), Flight_stage_1.y(tsecond_stage), Flight_stage_1.z(tsecond_stage), Flight_stage_1.vx(tsecond_stage), Flight_stage_1.vy(tsecond_stage), Flight_stage_1.vz(tsecond_stage), Flight_stage_1.e0(tsecond_stage), Flight_stage_1.e1(tsecond_stage), Flight_stage_1.e2(tsecond_stage), Flight_stage_1.e3(tsecond_stage), Flight_stage_1.w1(tsecond_stage), Flight_stage_1.w2(tsecond_stage), Flight_stage_1.w3(tsecond_stage)])
    #initial_solution = Flight_stage_1)
Flight_stage_2.plots.trajectory_3d()
````
Note that this does not include parachutes for simplicity's sake.

**Expected behavior**

I expected to see something like the following (minus when the rocket falls after it ejects the payload):

[image]

Which should be close to the example but just much higher: [https://docs.rocketpy.org/en/latest/user/deployable.html](url)

**Additional context**

Additional context:
I tried compensating for some of the positional initial conditions by specifying its new starting position in the environment block:
````
env2 = Environment(
    latitude= env1.latitude + (Flight_stage_1.y(tsecond_stage) / 6378000) * (180 / math.pi), # converted to degrees. This uses the equation: latitude = (latitude + y_coordinate/6378000)*(180/pi). 6378000 m is the Earth radius.
    longitude= env1.longitude + (Flight_stage_1.x(tsecond_stage) / 6378000) * (180 / math.pi) / (math.cos(env1.longitude * (math.pi / 180))), # again, a formula is used
    elevation=Flight_stage_1.z(tsecond_stage), # elevation must be updated to avoid a discontinuity
) 
env2.set_date((tomorrow.year, tomorrow.month, tomorrow.day, 12))
env2.set_atmospheric_model(type="Ensemble", file="GEFS")
````
This seemed to work fine, but it does not translate into the flight working.

Thanks for any help!

AndyBankes, 2025-03-05:
Maybe this could have been a problem with initial conditions?:
````
If a Flight object is used, the last state vector will be
            used as initial solution. If None, the initial solution will start
            with all null values, except for the euler parameters which will be
            calculated based on given values of inclination and heading.
            Default is None
````

Gui-FernandesBR, 2025-03-09:
- You can use the same environment obj for all the flights
- no need to "update" surface elevation in the second object 
- the initial solution can be optimzed in your code. Have you tried to simply pass the first flight as the initial solution? Or even using the flight.get_solution_at_time(t) method?
- when reporting a bug please do provide traceback and/or screenshots of what is wrongly happening in your code outputs, this makes response much easier

AndyBankes, 2025-03-14:
I have done it using env1 for both as well, however I had originally changed the environments in case this was a problem. Reverting the environment back and inputting initial_solution (as below) doesn't show any errors, it just takes an indefinitely long time to load ( after an hour I assumed something was wrong as it should not take that long to load).

[image]

This is why I'm not sure what I can be doing to fix this. 

One thing I was thinking may be the problem is related to the payload not having any input Thrust Curve (as it is a payload, not a rocket). I was reading a different issue from the issues subheading  [https://github.com/RocketPy-Team/RocketPy/issues/411](url) and noticed @williambrenner expected something a bit similar to me. Below is an image of what the trajectory looked like when I originally tried it:

[image]

Is something similar happening where it is trying to read a non-existent thrust curve and getting confused or something?

Thanks for your help Gui, do you have any other ideas on why it stalls like this?

MateusStano, 2025-03-15:
Hey! So it seems something in your input data is poorly dimensioned. For the simulation you've given, you are adding some big aerodinamic surfaces to your payload (`rocket_stage_2` as you called). Also, your drag for the payload seems a little high.

If you take off the surfaces and set both power on and power off drag to 0 the simulation runs fine.

I'd suggest taking a better look at those values

Furthermore, your simulation actually runs, it just stagnates eventually because of these poorly defined values, and so it doesn't reach a termination condition (i.e. impact point).

 If you want, you can have a closer look to what is happening by setting the `max_time` argument of the Flight class. In your simulation I set `max_time=98`, which then allowed me to take a closer look at what was happening

AndyBankes, 2025-03-27:
Thanks MateusStano.

Gui-FernandesBR, 2025-03-27:
Given it is not a library bug, rather a problem with the way the simulation was set, I'm closing this issue

closed 2025-03-27 by Gui-FernandesBR