RocketPy-Team/RocketPy #784: ENH: Create StochasticAirBrakes

kevin-alcaniz, 2025-03-15:
Currently, it’s not possible to account for air brakes control in the Monte Carlo simulations. This feature would be useful to validate the control algorithm under uncertainty.

I was thinking of developing a new class called StochasticAirBrakes, which would allow defining uncertainties in the drag curves. It will also be necessary to create a new function, add_air_brakes, in the StochasticRocket class, similar to the one in the deterministic class.

I’d like to work on this. What do you think?

LUCKIN13, 2025-03-17:
That would be an awesome idea, since many times, because of effect of the compression forces on the rocket and the pressure in the tongues. How are you thinking of adding the uncertainties? Adding a model of the airbrakes actuation would be something I would believe to be great with this feature.

kevin-alcaniz, 2025-03-18:
First, we need to add the capability to include an airbrake in the StochasticRocket, as this is currently not possible. Even without uncertainties, this would allow us to test how airbrake control works when accounting for rocket uncertainties.

Next, I would add the ability to introduce uncertainties to the airbrake itself. I believe the most important uncertainty in airbrakes is their drag curve, as validating these kinds of parameters before launch is always very challenging. To address this, I would add a drag_coefficient_curve_factor, similar to how it's done for "power on/off drag" in StochasticRocket. Other airbrake attributes, such as reference area and deployment level, could also be stochastic, but that’s not the main focus. I’d still implement it to maintain generality.

I've already developed this new feature. Would you like me to open a PR so you can review it?

kevin-alcaniz, 2025-03-18:
It's a pleasure to contribute! I already did the PR. Looking forward to getting your feedback!!

closed 2025-03-23 by Gui-FernandesBR