RocketPy-Team/RocketPy #787: Bug Report: Parachute AGL altitude doesn't account for given lag

L30-stack, 2025-03-19:
**Describe the bug**

_FlightPrints.events_registered returns the AGL altitude without lag delay.

**To Reproduce**

Add parachute with lag.

**Expected behavior**

"Altitude at inflation:" should print the altitude of parachute deployment including the delay.

Gui-FernandesBR, 2025-03-19:
As far as I can recall, the Flight.all_info() prints out both trigger and full deployment (with added lag) times.

IfnI understood correctly, your bug report tells us the altitude we are registering in the class is wrongly the altitude of trigger rather than the fully ejection.

L30-stack, 2025-03-19:
It's actually just a simple typo in the _FlightPrints.events_registered Function. 

`        def events_registered(self):
        """Prints out the Events Registered available about the flight.

        Returns
        -------
        None
        """
        print("\nParachute Events\n")
        if len(self.flight.parachute_events) == 0:
            print("No Parachute Events Were Triggered.")
        for event in self.flight.parachute_events:
            trigger_time = event[0]
            parachute = event[1]
            open_time = trigger_time + parachute.lag
            speed = self.flight.free_stream_speed(open_time)
            altitude = self.flight.z(open_time)
            name = parachute.name.title()
            print(f"Parachute: {name}")
            print(f"\tEjection time: {trigger_time:.3f} s")
            print(f"\tInflation time: {open_time:.3f} s")
            print(f"\tFreestream speed at inflation: {speed:.3f} m/s")
            print(
                f"\tAltitude at inflation: {altitude:.3f} m (ASL) | "
                f"{self.flight.altitude(**trigger_time**):.3f} m (AGL)"
            )`

`tigger_time` should be `open_time `.

Gui-FernandesBR, 2025-03-19:
> It's actually just a simple typo in the _FlightPrints.events_registered Function.
> 
> ` def events_registered(self): """Prints out the Events Registered available about the flight.
> 
> ```
>     Returns
>     -------
>     None
>     """
>     print("\nParachute Events\n")
>     if len(self.flight.parachute_events) == 0:
>         print("No Parachute Events Were Triggered.")
>     for event in self.flight.parachute_events:
>         trigger_time = event[0]
>         parachute = event[1]
>         open_time = trigger_time + parachute.lag
>         speed = self.flight.free_stream_speed(open_time)
>         altitude = self.flight.z(open_time)
>         name = parachute.name.title()
>         print(f"Parachute: {name}")
>         print(f"\tEjection time: {trigger_time:.3f} s")
>         print(f"\tInflation time: {open_time:.3f} s")
>         print(f"\tFreestream speed at inflation: {speed:.3f} m/s")
>         print(
>             f"\tAltitude at inflation: {altitude:.3f} m (ASL) | "
>             f"{self.flight.altitude(**trigger_time**):.3f} m (AGL)"
>         )`
> ```
> 
> `tigger_time` should be `open_time `.
> 
> I can do a PR later

Ok I agree, the last code line is wrong, we would kindly receive your PR

Gui-FernandesBR, 2025-03-19:
@L30-stack thank you for your contribution and quick responses!

closed 2025-03-19 by Gui-FernandesBR

L30-stack, 2025-03-19:
Thanks for being so easy about contributions!