RocketPy-Team/RocketPy #796: ENH: Add the Coriolis Acceleration

kevin-alcaniz, 2025-03-25:
I noticed that RocketPy currently doesn’t account for Coriolis effects in its equations.

I made a few local modifications to include this behavior and found that, for a rocket with a 3 km apogee, the deviation at the impact point is less than 10 meters.

Since these changes were minimal, I thought you might be interested in checking them out.

Let me know if you're interested!

Gui-FernandesBR, 2025-03-25:
@kevin-alcaniz great.
Yes, please do share with us the implementation you are proposing, and also let us check with you the deviations you've mentioned

kevin-alcaniz, 2025-03-26:
Do I make a pull request??

Gui-FernandesBR, 2025-03-26:
> Do I make a pull request??

Yes, please! Let it clear that your code is a code suggestion and we will discuss it thoroughly. 

Kevin, I added you as an external collaborator to this repo, this should make it possible for you to create new branches on this repository directly (rather than on your own fork). If you agree, I'd recommend you to start opening branches on this repo directly, whenever you desire to pull request new changes. This would make the review process easier, given that the whole organization will have easier access to your branch. 

In case you still want to keep pulling from your fork onto our develop branch, that's ok as well.

kevin-alcaniz, 2025-03-27:
Great!! I'm going to create the branch

kevin-alcaniz, 2025-03-27:
I've already uploaded my local branch to the repo. I'll submit the PR later, as I want to include a brief theoretical explanation to justify the changes. In the meantime, you can check it out!

closed 2025-07-26 by Gui-FernandesBR