RocketPy-Team/RocketPy #801: ENH: Enable only radial burning

caioessouza, 2025-04-04:
**Is your feature request related to a problem? Please describe.**

Rocketpy SolidMotor class only simulates radial and axial simoultaneos burn, but the regression rate on hybrid motor are modeled to be a function of the oxidizer mass flux entering the grain, so it only takes into consideration the radial burn.

**Describe the solution you'd like**

The idea is to implement the option to simulate only radial burn for the cases of a  axially inhibited grain or hybrid motors.

**Additional context**

[image]

[image]

Heeg, F., Kilzer, L., Seitz, R., & Stoll, E. (05 2020). Design and Test of a Student Hybrid Rocket Engine with an External Carbon Fiber Composite Structure. Aerospace, 7, 57. doi:10.3390/aerospace7050057

Gui-FernandesBR, 2025-04-04:
does it happen with **every** hybrid motor?

caioessouza, 2025-04-24:
> does it happen with **every** hybrid motor?

I can't ensure you that it happens to **every** hybrid motor. It's a model, as the combustion happens in the boundary layer between the liquid oxidizer and the solid fuel, the hypothesis is that the reaction is much more prominent in the port, where the burn is axial, because of the stronger oxidizer flow velocity, while the reaction in the bases of the grain are negligible.

This model is based on the source below and is largely cited, including in NASA SP7002-02.
G. A. MARXMAN, C. E. WOOLDRIDGE and R. J. MUZZY. Fundamentals of Hybrid
Boundary-Layer Combustion. In: Heterogeneous Combustion. Vol. 15. Progress in As-
tronautics and Rocketry.

caioessouza, 2025-05-02:
I runned some sanity check tests, just to see if the physical behaviour is correct, I used the motor Pro75M1670, defined in the getting started file.

[image]

1. I was expecting to the burn rate to grow in the begining, since it's calculated using
$r = - \frac{\dot m}{A_b \rho}$
and it was exactly what I got.

[image]

2. The second step was to analyse the burn area, and here is where the things started to get weird. I was expecting to both of the curves to end at the same time, the one with radial + axial burn being something similar to a parabola and the one with only radial burn being progressive. The tendecies were what I expected, but the radial burn only arrived at the final value earlier, what indicates that the burn finishes earlier.

[image]

3. To check better the behaviour, I plotted the inner radius evolution for both cases.

[image]

The results are coherent since   $\dot{R_i} = r$, but once this behaviour is leading to the grain to end while the motor still generates thrust, I'm having some problems to interpretate it.

4. The last sanity check I did was to plot the product of the burn rate with the burn area, since it should be constant to the same motor, and it actually is, so the problem is really in the evaluation of the burn area.

[image]

I will keep trying to interpret the results and correct the physical behaviour.

caioessouza, 2025-05-03:
Discovered the error, I forgot to change the grain_height_derivative, so, besides all te calculations were being done with only the port area, the height was still being consumed.

[image]

Now I just changed that and the behaviour is correct, so I will open the PR.

[image]
[image]
[image]
[image]
[image]