RocketPy-Team/RocketPy #804: BUG: MonteCarlo simulations not working when there are StochasticParachutes

Lucas-Prates, 2025-04-18:
**Describe the bug**

The `MonteCarlo.simulate` method encounters an `ValueError` when evaluation the pressure on the parachutes.

**To Reproduce**

Try to run a Monte Carlo simulation with a `StochasticParachute`. Otherwise, just run the _monte_carlo_class_usage_ notebook.

**Error debugging**

I debugged the code for a bit and the problem seems to be that the `time_overshoot` has a default of `None` in the `StochasticFlight` class, introduced in #785 . A `Flight` object is created with `time_overshoot=None`, which in turn makes the `__simulate` method skip running the `__calculate_and_save_pressure_signals` function, which in turn makes `parachute.clean_pressure_signal` and, finally, causes and error in `__transform_pressure_signals_lists_to_functions` since an empty argument is passed to a `Function`.

**Possible fix**

When I set `time_overshoot=True`, that works just fine. 

When I set `time_overshoot=False`, an `ValueError` is encountered once again, this time because several arrays related to contain NaN or Inf, possibly because of caching ? (See the screenshot)

An easy fix is to make `time_overshoot` equals of the base flight when it is None:

```
if time_overshoot is None:
    self.time_overshoot = flight.time_overshoot
else:
    self.time_overshoot = time_overshoot
```

However, that will not fix the problem when `time_overshoot` is `False` for the base flight.

**Screenshots**

Bug encountered in pressure evaluation (`time_overshoot=None`):
[image]

Bug encountered (`time_overshoot=False`)

[image]

**Additional context**

Shout-out to **MRT014 [LURA]**, a discord member that found the bug and provided me good feedback. Thank you :)

kevin-alcaniz, 2025-04-19:
Yes, this is the bug I mentioned in Issue #802!

I came to the same conclusion: the problem in the first part is that the default value for `time_overshoot` in the `StochasticFlight` class was set to `None`. It should match the base flight. I agree with that.

However, there’s still an issue in the `Flight` class—it doesn’t define some values when `time_overshoot` is set to `False`, which leads to a `KeyError` in the `MonteCarlo` class. This should be fixed to prevent further issues.

Gui-FernandesBR, 2025-04-24:
Great work guys,

@Lucas-Prates @kevin-alcaniz in my understanding the latest #805 PR have successfully solved this issue, am I wrong?

In that case I'd like to propose we close this issue.

kevin-alcaniz, 2025-04-25:
@Gui-FernandesBR Yes! The Monte Carlo is already working! Thank you!

PR #805 (2025-04-21): Bug: Fix StochasticFlight time_overshoot None bug
## Pull request type

- [x] Code changes (bugfix, features)
- [ ] Code maintenance (refactoring, formatting, tests)
- [ ] ReadMe, Docs and GitHub updates
- [ ] Other (please describe):

## Checklist

- [x] Lint (`black rocketpy/ tests/`) has passed locally 
- [ ] All tests (`pytest tests -m slow --runslow`) have passed locally
- [ ] `CHANGELOG.md` has been updated (if relevant)

## Current behavior

`time_overshoot=None` causes an error in `Flight._simulate`.

## New behavior

Fixes #804 by setting `time_overshoot` to the base flight value when [... cut]
closed 2025-04-25 by Gui-FernandesBR