RocketPy-Team/RocketPy #806: BUG: Paralle Monte Carlo simulation giving incorrect iteratiosn

emtee14, 2025-04-18:
When running the parallel Monte Carlo simulations, there is a printout that shows the current iterations, iteration runtime, and estimated time left. When running in parallel, though, the value goes up and down due to this line
https://github.com/RocketPy-Team/RocketPy/blob/4df0b383c2886c7804b6848bbe5b9cbe612b2cd5/rocketpy/simulation/monte_carlo.py#L431
```python
                    sim_monitor.print_update_status(sim_idx)
```

This means that the message is printed with the same index as the last completed iteration, so when an earlier iteration takes longer, it updates it with a lower value. 

I propose to fix this by having a counter that will keep track and just increment each time. This counter can be incremented when the output is written to a file, and can be thread-safe since it will only be accessed when the mutex is acquired.

**Screenshots**

https://github.com/user-attachments/assets/984e895b-01e0-453e-a127-061a7ab269f0

Gui-FernandesBR, 2025-04-24:
> I propose to fix this by having a counter that will keep track and just increment each time. This counter can be incremented when the output is written to a file, and can be thread-safe since it will only be accessed when the mutex is acquired. Also changing the printout of `current iteration` to iterations completed or something of that nature.

Thanks for reporting this bug, @emtee14 , it seems like you have a great idea of what should be done. Are you able to implement and test it on Python?
Could you kindly submit a Pull Request so we can evaluate your solution please? 

Thanks for helping!

Gui-FernandesBR, 2025-05-01:
> Yeah I'm happy to do that

Great!
@emtee14 I have assigned the issue to you then.
Let us know in case you need any help to proceed.

closed 2025-12-04 by Gui-FernandesBR